fossasia/pslab-app #2444: Wave Generator: phase cannot be selected for first waveform

marcnause, 2024-05-28:
**Actual Behaviour**

If Wave1 (analog wave) or SQR1 (digital wave) is selected, tapping the Phase-button has no effect.

**Expected Behaviour**

If this is due to hardware limitations, then the button should be disabled (greyed out).

If this is not due to hardware limitations, then it should be possible to select phase and change it via the seekbar.

**Steps to reproduce it**

1. Open app
2. Choose Wave Generator
3. Tap Wave1 button
4. Tap Phase button

or

1. Open app
2. Choose Wave Generator
3. Tap Digital button
4. Tap SQR1 button
5. Tap Phase button

**Would you like to work on the issue?**

Maybe

AsCress, 2024-05-29:
@marcnause As I can see in the code, the phase for Wave1 (analog wave) and SQR1 (digital wave) has purposely been disabled in the code. I am not sure if this is due to hardware limitations.

marcnause, 2024-05-29:
@AsCress If I find the time, I will dig through the [PSLab Python library](https://github.com/fossasia/pslab-python/tree/development) which is well documented to find out if it is a hardware limitation.

marcnause, 2024-05-29:
OK, I checked the Python library and the comments in https://github.com/fossasia/pslab-python/blob/development/pslab/instrument/waveform_generator.py were very enlightening:

`Phase between waveforms when generating waveforms on both SI1 and SI2 in degrees. The default is 0.`

The same is true for digital waveforms.

I think it would be a good to color the button grey when it is disabled. Currently it is not possible to tell apart if a button is bright red because it is not in active state or if it is disabled.

AsCress, 2024-05-30:
Right. I think that we should make the phase buttons invisible for the waveforms which don't support a phase change, to differentiate between _inactive_ and _disabled_. Does that sound right ? @marcnause @CloudyPadmal

marcnause, 2024-05-30:
@AsCress Sounds good to me.

PR #2452 (2024-06-01): fix: made the phase buttons invisible for Analog Wave 1 and Digital Wave SQR1
Fixes #2444 
Toggles the visibility of the phase buttons to set them to _INVISIBLE_ for Analog Wave 1 and Digital Wave SQR1 as a phase change for these waves isn't supported by our hardware.

## Changes 
- app/src/main/java/io/pslab/activity/WaveGeneratorActivity.java

## Screenshots / Recordings  
Here are screenshots showing the screen when Analog Wave 1 and Digital Wave SQR1 are selected.

[image]

[image]

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors [... cut]
closed 2024-06-01 by mariobehling