fossasia/pslab-app #2458: Oscilloscope: Add new feature for Position Controls

AsCress, 2024-06-03:
This is for the addition of the following proposed feature:-

**Horizontal & Vertical Position Controls**:
- Another feature would be to control the horizontal and vertical position of the waveform on the oscilloscope display, for manual measurements.
- The horizontal position control moves the waveform back and forth on the display and can be used to either align the waveform with the horizontal divisions of the display or to view different sections of the waveform.
- The vertical position control moves the waveform up and down on the display and can be used for manual measurements in a similar way as the horizontal position control.

**Actual Behaviour**
Currently, the app doesn't have any such feature to control the position of the waveform.

**Expected Behaviour**
Such a feature is an integral part of dedicated oscilloscope devices, and thus, our app should also be having it.

**Steps to reproduce it**
N/A

**LogCat for the issue**
N/A

**Screenshots of the issue**
N/A

**Would you like to work on the issue?**
Yes

AsCress, 2024-06-04:
[image]
@marcnause Just wanted to confirm is there any UI element, such as a button or so, which pops up in the area marked with the red rectangle, when the PSLab Device is connected ? I couldn't see any in the code, I think. Because, if there isn't any element there and the space is free, we can work on an even better layout for the Position Controls feature.

marcnause, 2024-06-04:
@AsCress You are right, the layout does not change when the PSLab device is connected.

[image]

PR #2471 (2024-06-12): feat: added Position Controls
Resolves #2458 
Adds the Position Controls feature to the Oscilloscope screen.

**Horizontal & Vertical Position Controls**:
- Another feature would be to control the horizontal and vertical position of the waveform on the oscilloscope display, for manual measurements.
- The horizontal position control moves the waveform back and forth on the display and can be used to either align the waveform with the horizontal divisions of the display or to view different sections of the waveform.
- The vertical position control moves the waveform up and down on the display an [... cut]