fossasia/pslab-app #2469: Enhance the Auto-Scale feature for Noise Detection

AsCress, 2024-06-05:
**Actual Behaviour**
Currently, the Auto-Scale feature just calculates the fundamental frequency of the incoming signal (which comes out to be arbitrary in the case of noise) and scales the axes on the basis of that.
Even if there is no signal, i.e. only noise is present, it would scale the axes on the basis of the arbitrary frequency measured of the noise.

**Expected Behaviour**
We can enhance the Auto-Scale feature to differentiate between _noise_ and _pure signal_ and only scale the axes if a _pure signal_ is detected. Else, a _Toast_ message saying "No input found", would be displayed. This is very akin to the feature that is already present in the dedicated oscilloscope devices.

**Steps to reproduce it**
1. Turn ON any channel of your choice.
2. Connect the inputs for the channel but don't supply any waveform, so that only noise is present.
3. Use the Auto-Scale feature.
4. One would observe that it scales the axes according to the noise.

**LogCat for the issue**
N/A

**Screenshots of the issue**
Here, is a recording showing the above steps with the In-Built MIC channel:-

https://github.com/fossasia/pslab-android/assets/125425881/9e576339-8bcd-4b54-b6f8-5fb503f8ae0d

**Would you like to work on the issue?**
Yes.

@marcnause Your comments on this ?

AsCress, 2024-06-06:
@marcnause I have developed this feature.
Here is a screen recording showing a demo with the In-Built MIC:-

https://github.com/fossasia/pslab-android/assets/125425881/9191546c-d7a4-4eb4-b00f-f59fc40b3a91

Although I won't be able to make a PR for this enhancement until we resolve all the issues with #2463 and merge that because this would cause merge conflicts, you can still test this with all the other channels if you want. The branch for this is _issue2469_ in my fork repo.

marcnause, 2024-06-06:
When I tested I got the exception in the following file: [logcat.txt](https://github.com/user-attachments/files/15686533/logcat.txt)

I can reproduce it if I select the in-build microphone first, press "AUTO", deselect the in-build microphone, select CH1 without any active wave, press "AUTO"

Also when I create a wave with the Wave Generator, change to the Oscilloscope and tap "AUTO" repeatedly, I see the Toast with the message even though the signal is no noise.

AsCress, 2024-06-07:
@marcnause Thanks so much for testing. Got your points. I'm working to improve this feature.

AsCress, 2024-06-08:
@marcnause I have fixed that exception and changed the algorithm for noise detection a bit. Can you please test it again ? The algorithm is based on some threshold values which I have currently set on the basis of the _In-Built MIC_. So, it may behave a bit different when used with the other channels. However, to adjust that, I have added appropriate logging that would help me adjust the threshold values. So when you test this, please enable logging and share the logcat file with me. Thanks:))

marcnause, 2024-06-09:
I tested again and it works very reliable now. 

There were a few  problems which were already fixed via https://github.com/fossasia/pslab-android/pull/2463 and I think you should create a pull request for this enhancement.

PR #2473 (2024-06-11): feat: added the Noise Detection enhancement to the Auto-Scale feature
Resolves #2469 
Adds the Noise Detection enhancement to the Auto-Scale feature.
The auto-scale feature would now be able to differentiate between _noise_ and _pure signal_, and would only scale the axes according to the input if a _pure signal_ is present.
## Changes 
- app/src/main/java/io/pslab/activity/OscilloscopeActivity.java
- app/src/main/java/io/pslab/communication/AnalyticsClass.java
- app/src/main/res/values/strings.xml

## Screenshots / Recordings  
Here is a demo of the feature-

https://github.com/fossasia/pslab [... cut]