fossasia/pslab-app #2470: Sine wave not generated correctly if digital waveform is used at the same time

marcnause, 2024-06-05:
**Actual Behaviour**

If an analogue waveform and a digital waveform is generated with the PSLab device at the same time, the analogue waveform is not generated as selected.

**Expected Behaviour**

If an analogue waveform and a digital waveform is generated with the PSLab device at the same time, both waveforms are generated as selected.

Or if this is a limitation of the PSLab hardware, it should not be possible to generate analogue and digital waveforms at the same time.

**Steps to reproduce it**

1. Connect Sine Wave 1 with CH1
2. Connect Square Wave 1 with CH3
3. Open app
4. Go to Wave Generator
5. Set frequency for analogue waveform
6. Set frequency for digital waveform
7. Start waveforms
8. Go to oscilloscope

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Maybe.

marcnause, 2024-06-05:
@bessman, do you know if this is a limitation of the PSLab hardware?

bessman, 2024-06-06:
This is partly a limitation in the hardware, partly in the firmware.

Channels SI2/SQ3 are both controlled by MCU pin 56, same for pins SI2/SQ3 which are controlled by MCU pin 57. This means that the hardware cannot generate a square wave on SQ4/SQ3 and simultaneously generate an analog wave on SI1/SI2.

However, the hardware __should__ be able to generate a square wave on SQ3 and simultaneously generate an analog wave on SI1. This does not currently work due to a firmware limitation; the firmware function responsible for generating square waves on SQ3 and SQ4 use certain registers which are also used by the analog waveform generator.

The hardware limitation should be communicated to the user in some way. I should be able to remove the firmware limitation, but until then it should also be communicated to the user.

marcnause, 2024-06-06:
Thank you for the clarification, @bessman!

I don't know what would be the best way to communicate the limitation to the user in an elegant way. Maybe change the UI to somehow indicate an analogue mode and a digital mode!?

marcnause, 2026-08-30:
TODO: Check if this is still valid for the PSLab v6 with the latest firmware.

closed 2026-09-02 by marcnause