fossasia/pslab-app #2472: PSLab v6 can't communicate with the Android device

AsCress, 2024-06-08:
**Actual Behaviour**
Due to the PSLab v6 device using a new Serial Bridge Controller (the CP2102N), the _CommunicationHandler.java_ currently doesn't support communication with the device.

**Expected Behaviour**
We should update the class to support serial communication with the PSLab v6 device, alongwith supporting backwards compatibility for the v5.

**Steps to reproduce it**
N/A

**LogCat for the issue**
N/A

**Screenshots of the issue**
N/A

**Would you like to work on the issue?**
@bessman @CloudyPadmal This is the issue ticket for coordinating on the progress.

bessman, 2024-06-08:
To summarize my findings so far:

The PSLab v5 uses a MCP2200 UART-USB bridge chip, while the v6 uses a CP2102N chip.

The MCP2200 uses CDC ACM (communications device class, abstract control model) to emulate a serial device. CDC ACM was meant to be a standard way to communicate with UART-USB bridges, but was not well received by the industry and MCP2200 is one of only a few chips that support it (in fact, the only one I've been able to find).

Most UART-USB bridges implement their own communication API. CP210x is such a device.

Linux, MacOS, and Windows have device drivers for both general ACM devices, and for the CP210x specifically. But Android lacks such drivers, meaning it is necessary to implement our own. CommunicationHandler currently implements an ACM driver.

For reference, we can look at the linux kernel's implementation: https://github.com/torvalds/linux/blob/master/drivers/usb/serial/cp210x.c

Linux is GPLv2 while pslab-android is Apache v2. But just looking at what bits linux sends and then doing something similar our own driver should be fine from a legal perspective.

I think it should be enough to send three commands to set up the connection:
1. IFC_ENABLE
2. SET_LINE_CTL with one stop bit, no parity, eight-bit word
3. SET_BAUDRATE to 1 Mbaud

I've tried this without success, but I suspect that I haven't properly disabled all the necessary parts of the ACM driver.

AsCress, 2024-06-08:
@bessman Thank you very much. I was also working on a similar approach. Did the app still crash when you made the changes ?

bessman, 2024-06-09:
Here is a copy of the v2.0 firmware, as discussed during today's meeting: [pslab-legacy-firmware.zip](https://github.com/user-attachments/files/15753337/pslab-legacy-firmware.zip)

AsCress, 2024-06-09:
> Here is a copy of the v2.0 firmware, as discussed during today's meeting: [pslab-legacy-firmware.zip](https://github.com/user-attachments/files/15753337/pslab-legacy-firmware.zip)

@bessman Thank you very much for this !!

PR #2488 (2024-06-23): feat: implemented serial driver to facilitate communication with PSLab Devices
Resolves #2472 
Implements the [usb-serial-for-android](https://github.com/mik3y/usb-serial-for-android) library to facilitate communication with the _PSLab_ device, as discussed in our weekly GSoC meet.

## Changes 
- This PR reconfigures the _CommunicationHandler.java_ class to implement the [usb-serial-for-android](https://github.com/mik3y/usb-serial-for-android) library, in order to facilitate communication with the new V6 device, while also having backwards compatibility with the V5.

## Screenshots / Recording [... cut]
closed 2024-06-23 by marcnause