fossasia/pslab-app #2492: Power Souce: EditTexts don't allow negative values for PV1 and PV2

marcnause, 2024-06-18:
**Actual Behaviour**

The values for PV1 and PV2 may be negative, but the `EditText`s for PV1 and PV2 don't allow to enter negative values.

**Expected Behaviour**

The `EditText`s should allow negative values for PV1 and PV2.

It is possible to allow negative decimal numbers as values by setting the `android:inputType` to `numberSigned|numberDecimal`, but this introduces problems due to a bug in the keyboards of older Android version (before Android O). A workaround is required. See https://stackoverflow.com/questions/3821539/decimal-separator-comma-with-numberdecimal-inputtype-in-edittext/51983392 for more information and hints.

**Steps to reproduce it**

1. Open app
2. Open Power Souce
3. Try to enter negative value for PV1 or PV2 with keyboard

**Would you like to work on the issue?**
Maybe

Dishebh, 2024-09-10:
Hey @marcnause , if the issue is still valid, can I check this out?

marcnause, 2024-09-10:
Yes, go ahead if you like! I will assign this issue to you.

closed 2024-09-20 by marcnause