fossasia/pslab-app #2499: Add pin layout screens for PSLab v6

marcnause, 2024-06-23:
**Actual Behaviour**

The app contains pin layout screens for PSlab v5.

**Expected Behaviour**

The app contains pin layout screens for PSlab v5 ans PSlab v6.

**Steps to reproduce it**

1. Start app
2. Press menu symbol (3 dots) in upper right
3. Menu entries lead to pin layout screens for PSlab v5 only.

**Would you like to work on the issue?**

Maybe.

Surajkumar5050, 2024-12-09:
I would like to work on this issue. Please assign it to me, and I'll start by analyzing the current implementation for PSLab v5 and adding support for PSLab v6 pin layout screens.

samruddhi-Rahegaonkar, 2024-12-19:
@Surajkumar5050 Are you working on this issue ? 
@AsCress If he is not working I would like to solve this issue.

Surajkumar5050, 2024-12-19:
Hi @samruddhi-Rahegaonkar , my exams are currently going on, so please feel free to work on this issue. You can ask @AsCress for this.

samruddhi-Rahegaonkar, 2024-12-19:
@AsCress Should I work on this issue?

AsCress, 2024-12-21:
@samruddhi-Rahegaonkar Sure! Go ahead.

marcnause, 2024-12-23:
Just in case you have not seen it yet, images of the PSLab v6 board can be found at https://github.com/fossasia/pslab-hardware

samruddhi-Rahegaonkar, 2024-12-24:
Thanks! @marcnause Was looking for it only.

samruddhi-Rahegaonkar, 2025-01-12:
@AsCress Started working on this issue. any Guidelines for this?

samruddhi-Rahegaonkar, 2025-01-12:
@AsCress which editor has been used for color map for v5 image ?

AsCress, 2025-01-12:
@samruddhi-Rahegaonkar How are you approaching this ?
I may not have an idea about the editor that was used for the color map. 
@CloudyPadmal Do you have an idea about this ?

samruddhi-Rahegaonkar, 2025-01-12:
@AsCress All the process will same as v5 just need to create new xml and java files for v6 and creating new color map for v6 and pin informations dialog box after clicking on each pin.

AsCress, 2025-01-12:
@samruddhi-Rahegaonkar That's great. And how would the user be switching between the pin layouts for the _v6_ and the _v5_ ?

samruddhi-Rahegaonkar, 2025-01-12:
@AsCress User can switch through `pslab_connectivity_menu.xml` we can add two more menus below v5 for v6.

AsCress, 2025-01-12:
> @AsCress User can switch through `pslab_connectivity_menu.xml` we can add two more menus below v5 for v6.

Yup. This would be nice to start with. Go ahead and implement whatever you find best.

CloudyPadmal, 2025-01-12:
> @samruddhi-Rahegaonkar How are you approaching this ?
> 
> I may not have an idea about the editor that was used for the color map. 
> 
> @CloudyPadmal Do you have an idea about this ?
> 
> 

I used Inkscape for the design

samruddhi-Rahegaonkar, 2025-01-18:
Hello,
I have to use the same colours in the color map which is given in the following pdf ?

[image]

AsCress, 2025-01-18:
@samruddhi-Rahegaonkar Yes. This looks nice !

samruddhi-Rahegaonkar, 2025-02-05:
@marcnause @AsCress I am done with the work. should I send PR ?

AsCress, 2025-02-05:
> [@marcnause](https://github.com/marcnause) [@AsCress](https://github.com/AsCress) I am done with the work. should I send PR ?

Go for it !

closed 2026-05-27 by rahul31124