fossasia/pslab-app #2509: Add Support for Wifi (ESP-01) Module in the Android App

AsCress, 2024-07-01:
**Actual Behaviour**
Currently, the only way the app can communicate with the _PSLab_ device is using a USB cable. There isn't any support for a wireless ( Wifi or a Bluetooth) connection with the device.

**Expected Behaviour**
We can implement support for a wireless connection with the _PSLab_ device using the _ESP-01_ module as suggested by @mariobehling.

**Steps to reproduce it**
N/A

**LogCat for the issue**
N/A

**Screenshots of the issue**
N/A

**Would you like to work on the issue?**
Yes.
@CloudyPadmal I see there has been past work on this feature (some past GSoC project as well). I haven't worked before with the ESP-01 module but it is something which I am looking forward to work and integrate in our app over the next few months. Can you please shed some light on this topic ? If there has been work on this before, then what is its current status in the Android app ?

CloudyPadmal, 2024-07-01:
Hi @AsCress, the idea we had earlier was to have the ESP module run in WIFI_AP mode, where PSLab acts as an access point. This way, a mobile device (or a laptop) can connect to PSLab and issue commands to get the response from PSLab. I think this is one way to do it, and there are probably a few other approaches we can explore.

AsCress, 2024-07-01:
@CloudyPadmal Right I got it. Was this implementation working when it was built first (I see some code that seems to implement this using _httpAsyncTask_) ? Should we continue on this path or start from scratch ? What are your views on this ?

CloudyPadmal, 2024-07-02:
If I recall, we could get the PSLab verion ID over WiFi. May be you can try the one we have right now and decide on which path to take (as the current work done on this is very little).

AsCress, 2024-07-03:
@CloudyPadmal Thank you very much for the information! That seems interesting. I'll surely try the current implementation as you mentioned. This also makes me curious about one thing: do we need any kind of support in the firmware for this to work ? If yes, then do we have it already, since you were able to get the version ID info ?

CloudyPadmal, 2024-07-03:
@bessman made the UART calls generic in a way that you can use either of the UART ports to communicate which we were not able to do before. ESP TX RX pins are wired to UART2 module and we can use that interface.

AsCress, 2024-07-04:
@CloudyPadmal Thank you so much for the information again ! I'll surely try this out and revert back to you.

PR #2561 (2025-01-18): feat: added support for wireless communication using ESP01
Fixes #2509 
Adds support for wireless communication with the _PSLab_ board using the ESP8266 ESP01 chip.

## Screenshots / Recordings  

[image]

https://github.com/user-attachments/assets/9e5120e7-1720-4409-bdb3-99928f33491b

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformat [... cut]
closed 2025-01-18 by marcnause