fossasia/pslab-app #2522: Multimeter: LA3 and LA4 dispaly wrong frequency values

marcnause, 2024-08-04:
**Actual Behaviour**

The frequency measurements of LA3 and LA4 are not correct in the Multimeter.

**Expected Behaviour**

The Multimeter always displays the correct value for the frequency on all channels.

1. Connect Sine Wave 1 to LA3 or LA4
2. Connect PSLAb to phone
3. Open app (if it has not opened automatically
4. Set frequency for Sine Wave 1
5. Change to Multimeter
6. Select LA3 or LA4

**Would you like to work on the issue?**

Maybe

AsCress, 2024-08-10:
@marcnause Could please share screenshots showing what exactly the wrong values of the frequency that you are getting on channels LA3 and LA4 ?

marcnause, 2024-08-10:
I set the frequency of `Sine Wave 1` to 1000 Hz and connected it to `LA1` first, then to `LA2`, then to `LA3`, and then to `LA4`. Below are the screenshots of the measurements. (PSLab v5, firmware 3.0.2)

[image]

[image]

[image]

[image]

PR #2560 (2024-11-03): fix: fixed Logic Analyzer
After months of work, this is finally here 🥳 !!
Completely fixes the _Logic Analyzer_ in our app.
Fixes #2522 

## Changes 
- The logic for generating timestamps based upon the data read from the _PSLab_ is now fixed.
- Two-Channel LA Capture (`CaptureTwo`) is now fixed by using the appropriate datatype `long`, for the data analysis.
- Three-Channel LA Capture has also been fixed by making it akin to the implementation in _PSLab Python_, i.e., starting a four channel capture and discarding the last channel.
- Frequency readings using the _Multimeter_ have also been fi [... cut]
closed 2024-11-03 by marcnause