fossasia/pslab-app #2523: Logic Analyzer: results on all 4 channels are shown even though only one channel is physically connected

marcnause, 2024-08-04:
**Actual Behaviour**

The Logic Analyzer displays results for all 4 channels, even though only LA1 is physically connected to a source.

**Expected Behaviour**

The Logic Analyzer only shows a result for channels which are connected to a source.

**Steps to reproduce it**

1. Connect LA1 to Square Wave 4
2. Open app (if it has not opened automatically
3. Select Wave Generator
4. Set frequency and duty cycle for Square Wave 4
5. Press "play" symbol on top of screen
6. Select Logic Analyzer
7. Change Channel Selection to 4
8. Press "Analyze" button

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Maybe

marcnause, 2024-08-11:
I tested again. I selected 4 channels.

If connected to `LA1` I get the same result as described above.
If connected to `LA2` I get a result for `LA2` only.
If connected to `LA3` I get a result for `LA3` only.
If connected to `LA4` I get no result.

I also tried with 3 channels.
If connected to `LA1` I get results on `LA1`, `LA2` and `LA3`
If connected to `LA2` I get a result for `LA2` only.
If connected to `LA3` I get no result.

AsCress, 2024-08-12:
@marcnause Thanks for testing ! These are exactly the kind of outputs that I'm getting with the V6.

closed 2024-11-03 by marcnause