fossasia/pslab-app #2524: Multimeter: Texts overlap and are partially cut off in horizontal orientation on some phones

marcnause, 2024-08-04:
**Actual Behaviour**

Texts overlap and are partially cut off in horizontal orientation in the Multimeter on some phones in horizontal orientation

**Expected Behaviour**

Texts don't overlap and are not cut off.

**Steps to reproduce it**

1. Open app
2. Go to multimeter
3. Chnage to horizontal orientation

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Maybe

AsCress, 2024-08-07:
@marcnause I think we'd have to design a separate layout for the multimeter for horizontal orientation as we have done for the other screens. Are you working on this or should I take this up ?

marcnause, 2024-08-07:
I will assign issues to myself once I start working on them or have a clear idea about what exactly needs to be done. Please feel free to work on any of the issues I reported which I have not assigned to myself.

AsCress, 2024-08-08:
@marcnause Thanks !! Sure. I'll take this up then and maybe create a new horizontal layout for the _Multimeter_ screen, as we've done for the other screens.

closed 2025-05-27 by marcnause