fossasia/pslab-app #2525: Power Source: output of PV1 and PV2 are not changed if set in app

marcnause, 2024-08-04:
**Actual Behaviour**

PV1 and PV2 always stay at the initial value, even if the value is set in the Power Source screen 

**Expected Behaviour**

PV1 and PV2 always output the voltage which is set in the Power Source screen

**Steps to reproduce it**

1. Connect PV1 (or PV2) to CH1
2. Connect PSLab to phone
3. Open app (if it has not opened automatically
4. Go to Power Source
5. Set value of PV1 (or PV2) a value different from the initial value
6. Go to Multimeter
7. Switch to CH1

I also measured with a hardware multimeter and git the same values.

**Additional information**
I tested with PSLab v5 with firmware version 3.0.2.

AsCress, 2024-08-11:
@marcnause Could you please share the log files when you are trying to set the _Power Source_ values ?

marcnause, 2024-08-11:
This is the log file: [logcat.txt](https://github.com/user-attachments/files/16576161/logcat.txt)

I flashed the v3.0.2 firmware build from https://github.com/fossasia/pslab-firmware/releases before testing again, but the result was the same as described previously.

AsCress, 2024-08-12:
@marcnause Thank you very much for providing the logs. I'll test this feature thoroughly and fix this soon in a PR.

PR #2541 (2024-08-24): fix: fixed Power Source
Fixes #2525 

## Changes 
- Issue #2525 reported by @marcnause was arising due to support for V5 **hardware** being removed mistakenly in 94ff43e9e86b44c8ec17fd60ae3a3f59324f712a merged in PR #2510.
- The change is now reverted.

## Screenshots / Recordings  
NA

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reforma [... cut]
closed 2024-08-24 by marcnause