fossasia/pslab-app #2526: Power Source: Unit mA cut off on devices with smaller screens

marcnause, 2024-08-04:
**Actual Behaviour**

The unit mA in the EditText of PCS is cut off in horizontal mode. (Test device: Moto X Play)

**Expected Behaviour**

Units are always fully visiblel.

**Steps to reproduce it**

1. Start app
2. Go to Power Source 
3. Change to horizontal orientation

**Screenshots of the issue**

[image]

[image]

Dishebh, 2024-09-10:
Hey @marcnause , does the issue still exist? I tried reproducing it in various dimensions in landscape mode, but couldn't see the issue. Could I be missing something?

marcnause, 2024-09-10:
@Dishebh I tested with my Moto X Play and I still see the cut off unit for PCS ("m" instead of "mA").

I have created a hardware profile which can be used to create a virtual device which has the same screen as the real hardware:
[moto_x_play.zip](https://github.com/user-attachments/files/16950323/moto_x_play.zip)

I had to compress the file since it is not possible to attach XML files here.

The file must be unzipped and can be imported in the Virtual Device Configuration via the "Import Hardware Profiles" button.

[image]

When I run PSLab in the virtual device, go to the Power Source and turn the device right (Command + R), I get the same result as with the real hardware.

[image]

Dishebh, 2024-09-15:
Thanks a lot @marcnause , will check this out!

Dishebh, 2024-09-29:
Hey @marcnause , I was able to reproduce the issue. I can see it happening for values that may come in 4 decimal places, or when the resolution is low. Just to understand, what's the best way to solve this? Should the numbers get resized automatically as and when they are increased, so that every digit of the number is always visible?

marcnause, 2024-09-30:
> Should the numbers get resized automatically as and when they are increased, so that every digit of the number is always visible?

Yes, if that's possible, I think it would be a good solution.

marcnause, 2025-07-11:
The Java app will be replaced with the Flutter version which is currently in development. I will close this issue to avoid unnecessary work.