fossasia/pslab-app #2544: Translations need attention

marcnause, 2024-08-30:
**Actual Behaviour**

Not al strings are translated and some strings which are translated are identical for all languages.

**Expected Behaviour**

All strings should be translated and non-translateable strings should not be repeated in all translations, but be marked as such in default striings.xml.

**Would you like to work on the issue?**

Yes, if nobody else wants to do it.

Juhi-Lakdawala, 2024-09-03:
Hey I would like to work on this issue. Can you assign me

anuPhoenixbis, 2024-09-14:
I would like to work on this issue would you please assign me?

marcnause, 2024-09-15:
> I would like to work on this issue would you please assign me?

@Juhi-Lakdawala are you working on this issue or shall I assign it to @anuPhoenixbis?

Juhi-Lakdawala, 2024-09-19:
I am trying to resolve this issue currently I will inform you about the updates in a few days

Juhi-Lakdawala, 2024-09-30:
hey can you reassign me again

anileshwar12, 2024-10-14:
I would like to work on this issue would you assign me?

Razorquake, 2024-11-10:
Is this issue still open? If so, would you assign it to me?

marcnause, 2024-12-01:
@Juhi-Lakdawala, are you still working on this issue or should we assign it to somebody else?

SidW111, 2024-12-05:
hey i would like to work on this issue can u please assign it to me @marcnause please if the current assignee is not woorking the can you please assign it to mee

marcnause, 2024-12-16:
> I would like to work on this issue would you assign me?

@anileshwar12, would you still like to work on this issue?

anileshwar12, 2024-12-17:
Hi @marcnause , yes, I’d like to work on this issue,

marcnause, 2024-12-17:
No special guidelines. I only noticed the points reported in the first post of this issue and thought that it could be improved. There are some useful hints in https://developer.android.com/guide/topics/resources/string-resource.

ChehakTrehan, 2025-04-02:
Hii @marcnause is this issue still open? I would love to work on this please assign me
Thanks

AsCress, 2025-04-02:
> Hii @marcnause is this issue still open? I would love to work on this please assign me
> Thanks

@ChehakTrehan Go ahead and make a PR. That's the best way to claim an issue.

closed 2025-04-06 by marcnause