fossasia/pslab-app #2557: Oscilloscope: Add Support for the Right Y-axis

AsCress, 2024-10-08:
**Actual Behaviour**
Currently, the oscilloscope plots waveforms on all channels using a single Y-axis (the _left Y-axis_). Though we have the UI in place for another reference Y-axis (the _right Y-axis_), programmatic support for it isn't implemented.

**Expected Behaviour**
We could have support for both the axes (as is present in many dedicated oscilloscope devices).
This would open up the possibilities for implementing many new features (such as different scales for different channels and comparisons between waveforms).

**Steps to reproduce it**
N/A

**LogCat for the issue**
N/A

**Screenshots of the issue**
N/A

**Would you like to work on the issue?**
Maybe.

Shresth-Deorari, 2024-10-14:
Hello @AsCress, I am new to Open Source. Can you please assign me this issue.

AsCress, 2024-10-15:
Hey there @Shresth-Deorari. Welcome to Open Source !
This issue in general requires a good knowledge of Android Programming and testing with the _PSLab_ Board (though we're always here to do that for you, if you don't have a _PSLab_ Board).
If you're all set and good to go, I'll assign you this issue.
Happy Coding !