fossasia/pslab-app #2580: Crash in XY plot of Oscilloscope screen

marcnause, 2024-12-01:
**Actual Behaviour**

The app crashes under certain circumstances in the XY plot of the Oscilloscope screen.

**Expected Behaviour**

The app never crashes.

**Steps to reproduce it**

1. connect PSLab to phone
2. start PSLab app
3. open Oscilloscope screen
4. enable "in-Build MIC"
5. go to XY Plot
6. enable XY Plot checkbox
7. choose MIC1 in left spinner
--> Crash

**LogCat for the issue**
<pre>FATAL EXCEPTION: AsyncTask #3
Process: io.pslab, PID: 7838
java.lang.RuntimeException: An error occurred while executing doInBackground()
	at android.os.AsyncTask$4.done(AsyncTask.java:399)
	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
	at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
	at java.util.concurrent.FutureTask.run(FutureTask.java:271)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:289)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:919)
Caused by: java.lang.IndexOutOfBoundsException: Index: 4, Size: 4
	at java.util.ArrayList.get(ArrayList.java:437)
	at io.pslab.communication.ScienceLab.fetchData(ScienceLab.java:449)
	at io.pslab.communication.ScienceLab.fetchTrace(ScienceLab.java:415)
	at io.pslab.activity.OscilloscopeActivity$XYPlotTask.doInBackground(OscilloscopeActivity.java:1309)
	at io.pslab.activity.OscilloscopeActivity$XYPlotTask.doInBackground(OscilloscopeActivity.java:1267)
	at android.os.AsyncTask$3.call(AsyncTask.java:378)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:289) 
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
	at java.lang.Thread.run(Thread.java:919) </pre>

**Would you like to work on the issue?**

Maybe.

Surajkumar5050, 2024-12-09:
I would like to work on this issue.
I don't see any option "MIC1" in both spinners. There is 1 option "MIC" & it is not crashing
[image]

AsCress, 2024-12-09:
> I would like to work on this issue.
> 
> I don't see any option "MIC1" in both spinners. There is 1 option "MIC" & it is not crashing
> 
> [image]
> 
> 
> 
> 
> 
> 

@Surajkumar5050 Welcome here!
The first step in the "Steps to reproduce" says to connect the PSLab device with the phone.
Just wanted to confirm, did you connect the device and then check ?

Surajkumar5050, 2024-12-09:
@AsCress , I don't currently have the PSLab device to test the issue physically. However, after analyzing the code, I believe the issue might be caused by an `IndexOutOfBoundsException` in the `fetchData` method, particularly when accessing `this.aChannels.get(channelNumber - 1)`.

**Possible Fix:**
The exception occurs because the `channelNumber` exceeds the size of `aChannels`. I have modified the `fetchData` method in ScienceLab.java class to include additional validation:

```
private boolean fetchData(int channelNumber) {
    int samples = this.aChannels.get(channelNumber - 1).length;
    if (channelNumber > this.channelsInBuffer) {
        Log.v(TAG, "Channel Unavailable");
        return false;
    }
    Log.v("Samples", "" + samples);
    Log.v("Data Splitting", "" + this.dataSplitting);
    ArrayList<Integer> listData = new ArrayList<>();
    try {
        for (int i = 0; i < samples / this.dataSplitting; i++) {
            mPacketHandler.sendByte(mCommandsProto.COMMON);
            mPacketHandler.sendByte(mCommandsProto.RETRIEVE_BUFFER);
            mPacketHandler.sendInt(this.aChannels.get(channelNumber - 1).bufferIndex + (i * this.dataSplitting));
            mPacketHandler.sendInt(this.dataSplitting);
            byte[] data = new byte[this.dataSplitting * 2 + 1];
            mPacketHandler.read(data, this.dataSplitting * 2 + 1);
            for (int j = 0; j < data.length - 1; j++) {
                listData.add((int) data[j] & 0xff);
            }
        }

        if ((samples % this.dataSplitting) != 0) {
            mPacketHandler.sendByte(mCommandsProto.COMMON);
            mPacketHandler.sendByte(mCommandsProto.RETRIEVE_BUFFER);
            mPacketHandler.sendInt(this.aChannels.get(channelNumber - 1).bufferIndex + samples - samples % this.dataSplitting);
            mPacketHandler.sendInt(samples % this.dataSplitting);
            byte[] data = new byte[2 * (samples % this.dataSplitting) + 1];
            mPacketHandler.read(data, 2 * (samples % this.dataSplitting) + 1);
            for (int j = 0; j < data.length - 1; j++) {
                listData.add((int) data[j] & 0xff);
            }
        }

    } catch (IOException e) {
        e.printStackTrace();
        return false;
    }

    for (int i = 0; i < listData.size() / 2; i++) {
        if (i * 2 + 1 < listData.size()) {
            this.buffer[i] = (listData.get(i * 2)) | (listData.get(i * 2 + 1) << 8);
            while (this.buffer[i] > 1023) this.buffer[i] -= 1023;
        } else {
            Log.e(TAG, "Index out of bounds: " + (i * 2 + 1));
        }
    }

    Log.v("RAW DATA:", Arrays.toString(Arrays.copyOfRange(buffer, 0, samples)));

    this.aChannels.get(channelNumber - 1).yAxis = this.aChannels.get(channelNumber - 1).fixValue(Arrays.copyOfRange(this.buffer, 0, samples));
    return true;
}
```

Replace the existing **fetchData** method in the **ScienceLab.java** class with the code above.
Rebuild the project and run the application with the **PSLab device** connected to verify if the issue persists.

AsCress, 2024-12-09:
@Surajkumar5050 Sounds nice ! Would you like to give it a try and fix this ?

AsCress, 2024-12-09:
No issues at all! You can make the PR and we are more than happy to test it for you.

Surajkumar5050, 2024-12-09:
I have created the pull request, and it is now ready for testing. Feel free to proceed with the testing, and I'm happy to address any further issues if they arise.

marcnause, 2025-08-10:
Closing this issue since it concerns the old Java version of the app.