fossasia/pslab-app #2584: Not all sensors which have been implemented are shown in SensorActivity

marcnause, 2024-12-16:
**Actual Behaviour**

SensorActivity does not list all sensors. (9 sensors are listed.)

**Expected Behaviour**

SensorActivity should list all available sensors. (11 Actvities for sensors exist.)

The reason why not all sensors are listed is probably that there are duplicate keys in `sensorAddr` in `SensorActivity`.

**Steps to reproduce it**

1. Open app
2. Tap "Sensors"
3. Tap "AUTOSCAN" button

**LogCat for the issue**

N/A

**Screenshots of the issue**

[image]
[image]

**Would you like to work on the issue?**

Maybe.

samruddhi-Rahegaonkar, 2024-12-17:
Hello, I would Like to work on this issue

marcnause, 2024-12-17:
@samruddhi-Rahegaonkar I have assigned the issue to you. I am not sure what the numbers which are used as keys in in `sensorAddr` in `SensorActivity` mean.

@AsCress You probably added the new sensor implementations during GSoC. Do you remember where the numbers come from?

samruddhi-Rahegaonkar, 2024-12-18:
@marcnause I am Facing some issues while setting up the project . Why we don't have following code snippet for jitpack in build.gradle.

	dependencyResolutionManagement {
		repositoriesMode.set(RepositoriesMode.FAIL_ON_PROJECT_REPOS)
		repositories {
			mavenCentral()
			maven { url 'https://jitpack.io' }
		}
	}

samruddhi-Rahegaonkar, 2024-12-18:
@marcnause I am getting following error while building project do you know how to resolve it ?
[Fatal Error] jetifier-core-1.0.0-beta10.pom:2:90: The markup in the document following the root element must be well-formed.

samruddhi-Rahegaonkar, 2024-12-18:
@marcnause  Solved !👍

samruddhi-Rahegaonkar, 2024-12-18:
@marcnause Hi, I have submitted a PR #2589 that resolves this issue. Please let me know if further changes are needed. Thanks!

AsCress, 2025-01-11:
Completed in #2589.

closed 2025-01-11 by AsCress