fossasia/pslab-app #2587: App crashes if unkonwn host is entered

marcnause, 2024-12-17:
**Actual Behaviour**

The app crashes if the user enters an invalid host address.

**Expected Behaviour**

The app never crashes due to user input.

**Steps to reproduce it**

1. Start app
2. Choose "Connect Device" in the Navigation Drawer
3. Tap "WIFI" button
4. Enter ".lo"
5. Tap "CONNECT" button

**LogCat for the issue**

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

Yes.

samruddhi-Rahegaonkar, 2024-12-18:
Issue is Still Encountering.

samruddhi-Rahegaonkar, 2024-12-18:
@marcnause Should I check ? Assign this to me please.

marcnause, 2024-12-21:
> @marcnause Should I check ? Assign this to me please.

There is a pending pull request: https://github.com/fossasia/pslab-android/pull/2588

I tried to add you as a reviewer, but that did not work. Please feel free to test the automatically generated APK and comment the PR nevertheless.

PR #2588 (2024-12-22): fix: prevent crash if invalid host is entered
Fixes #2587

## Changes 
- catch exception if unresolveable host is entered
- use try-with-resources for autocloseable Response object
- log errors instead of writingt them to StdOut
- use lambdas instead of anonymous listener classes
- prevent unnecessary cast by using correct type (LayoutParams)

## Screenshots / Recordings  
N/A

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource f [... cut]
closed 2024-12-22 by marcnause