fossasia/pslab-app #2591: Double indication of the time units in the sensor screens

marcnause, 2024-12-22:
**Actual Behaviour**

In most sensor screens time units are indicated as "sec" and "s".

**Expected Behaviour**

Time units should only be indicated once per diagram and they should be the same everywhere (either "sec" or "s").

**Steps to reproduce it**

1. Open App
2. Tap "Sensors" card
3. Tap "AUTOSCAN" button
4. Select sensor, e.g. APDS9960

**Screenshots of the issue**
[image]

**Would you like to work on the issue?**

Maybe

ChehakTrehan, 2024-12-26:
Hey @marcnause can you please assign me this issue and let me know when you're available for a quick meet. This is my first contribution and had some questions. I hope you could help in resolving them. Thanks a lot! 

Best Regards
Chehak

marcnause, 2024-12-26:
I have assigned you this ticket, but unfortunately I will not be available for a meet today due to family obligations. I don't know when I will be available in the next few days yet. In which timezone are you located? 

Please feel free to ask questions here and I will answer them whenever I find some time.

VishnuSanal, 2025-02-17:
hi, I would like to work on this issue.

AsCress, 2025-02-17:
@VishnuSanal Go ahead and feel free to make a PR when you're ready.

PR #2633 (2025-02-27): fix: double indication of the time units in the sensor screens
Fixes #2591

## Changes 
- remove redundant textviews with text "(s)"
- remove unused string resource ("strings/seconds")
- replaced `time_space` string with `time`

## Screenshots / Recordings
[image]

**Checklist**:
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and fixed ind [... cut]
closed 2025-02-27 by AsCress