fossasia/pslab-app #2595: Scan for sensors always returns list of all sensors

marcnause, 2024-12-25:
**Actual Behaviour**

Scan for sensors always returns list of all sensors, even if no sensor or no PSLab boars is connected.

**Expected Behaviour**

Scan for sensors only returns sensors which are connected to PSLab.

**Steps to reproduce it**

1. Open app
2. Tap "Sensors"
3. Tap "AUTOSCAN"

**Screenshots of the issue**

[image]

**Would you like to work on the issue?**

No, @AsCress volunteered to take care of this issue. 😉

AsCress, 2024-12-28:
@marcnause The issue seems to be in the firmware and not in the Android app or the python library (I guess).
I created an issue ticket in the firmware repository (https://github.com/fossasia/pslab-firmware/issues/182).

marcnause, 2024-12-28:
OK, I will close this issue since it is not caused by the app. Thank you for investigating and for creating the ticket in the firmware repository.

closed 2024-12-28 by marcnause