fossasia/pslab-app #2598: Add support for MAX30100 Pulse Oximeter and Heart-Rate Sensor

marcnause, 2024-12-25:
**Actual Behaviour**

MAX30100 is not supported.

**<s>Expected</s> Desired Behaviour**

MAX30100 is supported. The MAX30100 is a pulse oximeter and heart-rate sensor. I think it would be a fun experiment to monitor body functions of the user.

Data sheet of the sensor: https://www.analog.com/media/en/technical-documentation/data-sheets/MAX30100.pdf

**Would you like to work on the issue?**

Yes, once I have the sensor. I may require support.

HITEE25, 2025-01-28:
can you help me to start contributing , I  am not understanding how to contribute

marcnause, 2025-01-28:
I don't think this is a good issue for a first contribution to the app since it requires a good understanding of how sensors communicate with the app.

As far as I know the following steps are required:
1. Create class which communicates with the sensor (requires info from data sheet)
2. Create screen to display data to user
3. Integrate new classes into app (new sensor needs to be added to SensorActivity)