fossasia/pslab-app #2614: Bluetooth: app crashes when conneceting to arbitrary device

marcnause, 2025-01-18:
**Actual Behaviour**
The app crashes when user tries to connect to a Bluetooth device.

**Expected Behaviour**
The app shouldn't crash in any circumstances.

**Steps to reproduce it**
Scan Bluetooth devices and try to connect to any device.

**LogCat for the issue**

> FATAL EXCEPTION: main
Process: io.pslab, PID: 18312
java.lang.RuntimeException: Error receiving broadcast Intent { act=android.bluetooth.device.action.FOUND flg=0x10 pkg=io.pslab (has extras) } in io.pslab.fragment.BluetoothScanFragment$1@5a2bfd
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1800)
	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7932)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:942)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.content.Context.checkPermission(java.lang.String, int, int)' on a null object reference
	at androidx.core.content.ContextCompat.checkSelfPermission(ContextCompat.java:552)
	at io.pslab.fragment.BluetoothScanFragment$1.onReceive(BluetoothScanFragment.java:53)
	at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0$android-app-LoadedApk$ReceiverDispatcher$Args(LoadedApk.java:1790)
	at android.app.LoadedApk$ReceiverDispatcher$Args$$ExternalSyntheticLambda0.run(Unknown Source:2) 
	at android.os.Handler.handleCallback(Handler.java:942) 
	at android.os.Handler.dispatchMessage(Handler.java:99) 
	at android.os.Looper.loopOnce(Looper.java:201) 
	at android.os.Looper.loop(Looper.java:288) 
	at android.app.ActivityThread.main(ActivityThread.java:7932) 
	at java.lang.reflect.Method.invoke(Native Method) 
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548) 
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:942) 

**Screenshots of the issue**

https://github.com/user-attachments/assets/441a64b0-994c-4a37-9b9d-1c07f191a83d

**Would you like to work on the issue?**
Maybe

AsCress, 2025-01-24:
@marcnause I guess this arises because of the fact that we haven't configured support for _Bluetooth_ connections in the app. I wonder whether we should work on issues relating to _Bluetooth_ ? We haven't decided yet if we should work on developing this feature in the app, provided that we've already implemented wireless connection using _Wifi_.
This is something which can be decided upon.

marcnause, 2025-01-26:
@AsCress As discussed in our weekly meeting, we should remove the Bluetooth option since Bluetooth is not supported, nobody is working on it and we have working WiFi support now.

marcnause, 2025-07-11:
The Java app will be replaced with the Flutter version which is currently in development. I will close this issue to avoid unnecessary work.