fossasia/pslab-app #2636: More frequent releases

froscon2024, 2025-03-04:
More frequent releases of the app would be great. The last release of the app is more then a half year old and i seem to have issues that are fixed since then.
Please push more frequent updates. A good known app-update value is every month one release up to at least one release every 2 months.

AsCress, 2025-03-04:
@froscon2024 Thanks for those comments! We'd take note of that. Meanwhile you could join the [beta](https://play.google.com/apps/testing/io.pslab) for always staying on the newest changes.

froscon2024, 2025-03-04:
Thanks for the quick answer. Joining some beta on google servers is impossible for me because i do not have any google account and i do not plan to ever get one.
I opened following issue based on the new information, that there are beta releases not available for people that do not use google accounts: https://github.com/fossasia/pslab-android/issues/2638

closed 2025-04-08 by froscon2024