fossasia/pslab-app #2637: Create some basic check when pslab board does not respond

froscon2024, 2025-03-04:
https://github.com/fossasia/pslab-python is reporting correctly a error that the pslab device does not respond and that why its reported as not detected.
The android app fails this basic test. The pslab firmware release 3.1.0 is broken on actual v6 hardware. It does not respond over serial. The python tool correctly reports that. The android app instead is missing this check. I had to spend much time trying to understand what is wrong and why nothing is working.
Such basic check should be implemented also in the android app.

thisisanubhav, 2026-09-22:
@rahul31124 @marcnause this has been assigned for a while with no linked PR. If you're not working on it, may I pick it up? My plan is to make the app send a basic handshake command once it connects over USB. If the board doesn't answer, it would show "device not responding" instead of treating it as connected, which matches what pslab-python does. I'll link a draft PR here; happy to drop it if you already have work in progress.

thisisanubhav, 2026-09-22:
Opened #3588. One correction to my plan above: the app already rejects a device that fails the version handshake, but it only logged it. So #3588 adds the missing part, a warning dialog telling the user the device didn't respond like a PSLab.

closed 2026-09-29 by marcnause