fossasia/pslab-app #2638: Add beta/testing releases to F-Droid and to github releases

froscon2024, 2025-03-04:
It looks like you have beta/testing releases of the android app for pslab. Sadly those are missing on F-Droid app-store.
Of course i am not using any google services, do not have a google account and i do not plan to ever have a google account. F-Droid have the ability to have both stable and testing releases. If a person just wants to install the app, then its choosing stable because of the suggested tag. If you want beta, then you can switch to beta.
Sadly at the moment you do not make use of this feature on f-droid: https://f-droid.org/en/packages/io.pslab/

Github also have such functionality to tag releases as testing and stable. You have even used it in this pslab repo: https://github.com/fossasia/pslab-hardware/releases

Please add the F-Droid and Github options for the people to be able to get beta/testing releases of this app without the need of any google services/accounts.

AsCress, 2025-03-04:
@froscon2024 Thanks so much for the information! I'll have a look at the beta track in FDroid soon, so that we may host the beta app on FDroid as well!

froscon2024, 2025-03-04:
Great. You can take a look for example on the F-Droid app itself: https://f-droid.org/en/packages/org.fdroid.fdroid/
There you can find regular alpha releases that of course never get the suggested tag on them. Only stable releases get the suggested tag.