fossasia/pslab-app #2639: Google Key is Exposed.

Wassaf001, 2025-03-05:
In gradle.properties the google key is exposed.

[image]

Shukla007, 2025-03-08:
can i work on this issue.

AsCress, 2025-03-18:
@CloudyPadmal Do you know anything about this ? Has the key been exposed since it was included in here ? Do we use it somewhere ?

Shukla007, 2025-03-18:
Yes, I have built the app locally, and it is hardcoded in the code. We can solve it by putting it into a .env file.

AsCress, 2025-03-18:
@Shukla007 That's right ! However, let's get some background on this first, if we do require the key or not. If not, we may just remove it instead.

Shukla007, 2025-03-18:
@AsCress Yes, you are correct. First, let me check the code to see if it uses the key or if it is just there without any purpose.

sanmoho123, 2025-03-19:
please assign me this issue i went through it and i came out with this thought, Remove the exposed API key from gradle.properties, store it securely in a secrets.properties file (ignored by Git), and load it dynamically in the build script to prevent unauthorized access and enhance security.

CloudyPadmal, 2025-03-21:
> [@CloudyPadmal](https://github.com/CloudyPadmal) Do you know anything about this ? Has the key been exposed since it was included in here ? Do we use it somewhere ?

I would suspect this to be a Google Maps key. I'm not sure if we are using it anymore. Maybe best to remove it and move forward.

sanmoho123, 2025-03-21:
> > [@CloudyPadmal](https://github.com/CloudyPadmal) Do you know anything about this ? Has the key been exposed since it was included in here ? Do we use it somewhere ?
> 
> I would suspect this to be a Google Maps key. I'm not sure if we are using it anymore. Maybe best to remove it and move forward.

Move the Google API key from gradle.properties to local.properties which is automatically excluded from version control.

Use the Secrets Gradle Plugin to securely access the key in your build process without exposing it in your repository.

AsCress, 2025-03-21:
> > [@CloudyPadmal](https://github.com/CloudyPadmal) Do you know anything about this ? Has the key been exposed since it was included in here ? Do we use it somewhere ?
> 
> I would suspect this to be a Google Maps key. I'm not sure if we are using it anymore. Maybe best to remove it and move forward.

@CloudyPadmal Perfect. That's the way forward. Let's just remove it.

sanmoho123, 2025-03-21:
> [@CloudyPadmal](https://github.com/CloudyPadmal) Perfect. That's the way forward. Let's just remove it.

yes, it would be best, Remove the exposed Google API key from gradle.properties and store it in local.properties which is git-ignored by default.

AsCress, 2025-03-21:
Yes. Anyone here would like to make a PR for the same ?

sanmoho123, 2025-03-21:
> Yes. Anyone here would like to make a PR for the same ?

yes sure i would like to do that

AsCress, 2025-03-21:
> > Yes. Anyone here would like to make a PR for the same ?
> 
> yes sure i would like to do that

Go ahead.

PR #2657 (2025-03-23): fix: removed unused Google Maps Key
Fixes #2639

## Changes 
- Removed Google Maps Key from gradle.properties

## Screenshots / Recordings  
[image]

**Checklist**: 
- [x] **No hard coding**: I have used resources from `strings.xml`, `dimens.xml` and `colors.xml` without hard coding any value.
- [x] **No end of file edits**: No modifications done at end of resource files `strings.xml`, `dimens.xml` or `colors.xml`.
- [x] **Code reformatting**: I have reformatted code and fixed indentation in every file included in this pull request.
- [x] **No extra space**: My code does not contain any extra  [... cut]
closed 2025-03-23 by AsCress