fossasia/pslab-app #2679: Phone Sensors: Screens are not designed for orientation change

marcnause, 2025-05-10:
**Actual Behaviour**
Recorded data is lost when the orientation is changed in on of the following screens: 
* Lux Meter
* Accelerometer
* Barometer
* Compass
* Gyroscope
* Thermometer
* Gas Sensor
* Dust Sensor
* Sound Meter

**Expected Behaviour**
Previously recorded data should still be visible after orientation change.

If possible, create/use base class to prevent duplicate code. (Check how it is done in the sensor screens.)

**Steps to reproduce it**
0. Open app
1. Open screen "Accelerometer"
2. Wait until data is visible in the graphs
3. Change orientation of screen

--> Data will be lost

**Screenshots of the issue**

https://github.com/user-attachments/assets/00514637-f409-41fe-a99f-5f20c6645f7d

**Would you like to work on the issue?**

Yes, but feel free to grab this issue as long as I have not assigned it to myself.

avinashgarlaa, 2025-07-03:
Hi! I'm interested in working on this issue as well. Could I be assigned if it's not already taken?

marcnause, 2025-07-11:
The Java app will be replaced with the Flutter version which is currently in development. I will close this issue to avoid unnecessary work.

closed 2025-07-11 by marcnause