fossasia/pslab-app #2685: The `reviewers` field in the `dependabot.yml` file will be removed soon. Please use the code owners file to specify reviewers for Dependabot PRs. For more information, see [this blog post](https://github.blog/changelog/2025-04-29-dependabot-reviewers-configuration-option-being-replaced-by-code-owners/).

marcnause, 2025-05-12:
The `reviewers` field in the `dependabot.yml` file will be removed soon. Please use the code owners file to specify reviewers for Dependabot PRs. For more information, see [this blog post](https://github.blog/changelog/2025-04-29-dependabot-reviewers-configuration-option-being-replaced-by-code-owners/).

_Originally posted by @dependabot[bot] in https://github.com/fossasia/pslab-android/issues/2682#issuecomment-2870884104_

Amitjoiya, 2025-11-16:
Hi! I inspected the `dependabot.yml` file as requested.

✔️ There is no `reviewers:` field present anywhere in the file.  
✔️ So no removal or modification is needed.  
✔️ I have not changed or reformatted any file to keep the PR clean, as instructed.

Please let me know if you want me to make any additional adjustments.

marcnause, 2025-11-21:
@Amitjoiya, thank you for having a look! I also haven't Seen the warning anymore. I guess we can close this issue.

closed 2025-11-21 by marcnause