fossasia/pslab-app #2692: String resources in Flutter App

CloudyPadmal, 2025-05-17:
Fetching string resources from a centralized constant file offers many advantages.

- Easy maintenance as all strings are stored in one place, making it easier to edit or update them without searching through multiple files.
- Improved localization support to transition to internationalization (i18n) using tools like the `intl` package in Flutter.
- Better organization separating UI strings from logic and other constants improves code readability and maintainability.
-  Consistency helps avoid typos and ensures consistency across the UI.

However, I’m not sure the current approach in the codebase fully aligns with these benefits.

Currently, strings are defined mixed with unrelated constants (such as numerical values).

I recommend wrapping them in a dedicated class.

```dart
class Strings {
  static const appName = 'PSLab';
  static const saveSuccessMessage = 'Saved successfully!';
  static const saveFailureMessage = 'Save failed. Please try again.';
}
```

Then in the UI

```dart
Text(Strings.saveSuccessMessage)
```

This pattern is both scalable and prepares the codebase for future i18n with minimal refactoring.

Also, the string resources are distinct from other variable names (with the `Strings.` prefix).

What are your thoughts, @AsCress @marcnause ?

AsCress, 2025-05-18:
@CloudyPadmal This is definitely a very good approach. I'll soon do a complete refactor of the _flutter_ app to follow this model :)

marcnause, 2025-05-19:
I found this: https://docs.flutter.dev/ui/accessibility-and-internationalization/internationalization

Would there be any reason against doing it this way?

AsCress, 2025-05-19:
> I found this: https://docs.flutter.dev/ui/accessibility-and-internationalization/internationalization
> 
> Would there be any reason against doing it this way?

@CloudyPadmal How do you like this approach?

CloudyPadmal, 2025-05-19:
That's the way to go. You still need a separate file with all the strings.

AsCress, 2025-05-19:
@CloudyPadmal Yeah exactly. Let me take care of this :)

marcnause, 2025-07-11:
Can we close this issue now that https://github.com/fossasia/pslab-android/pull/2779 is merged?

AsCress, 2025-07-12:
@marcnause Just added localization for German (automatically generated using a machine translator) and tested the app, everything works fine:

[image]
[image]

I'll make a pull request for this so that you can review the translations.

closed 2025-07-12 by CloudyPadmal