fossasia/pslab-app #2875: Create Gif files for experiment demonstration.

Yugesh-Kumar-S, 2025-08-24:
### Creating GIFs for Experiments

The GIFs used in the experiment guide screen were created using **Canva**:

1. Go to [https://www.canva.com](https://www.canva.com).
2. Create a new design ([card](https://www.canva.com/design?create&type=TAESO87vhFA&category=tAEwhcYByGs&analyticsCorrelationId=9d06fdc8-4bd9-4d9d-a32e-935676b19ef6) or custom dimensions as needed).
3. Add elements, animations, or screenshots to demonstrate the experiment steps.
4. Export the design as a GIF (Download → File Type → GIF).
6. Place the GIF file under `assets/images` and reference it in the experiment guide screens.

Yugesh-Kumar-S, 2025-08-24:
@marcnause I have documented the steps to create gif files for experiments here. Is this sufficient ?

marcnause, 2025-08-24:
👍 Thank you! I think it is sufficient. I'm not sure yet where to place the information so that it can be found by interested persons. I'll assign this ticket to myself and will think about it.